MatterVN/ModbusTCP2MQTT #1: Noob needs additional info

Yop1403, 2021-09-11:
Hi, have been seeking for an integration that supports Sungrow inverters, for quite a while (I've got the Sungrow SG5K).
Now my noob questions:
- do I need to install some extra MQTT integration?
- how/where do I define the login/password that are used by the ModbusTCP2MQTT integration?

Yop1403, 2021-09-15:
An additional note:
Is this integration really stable for all of you? It seems my inverter (Sungrow SG5K) goes into a sort of standby mode every night, the next day this integration won't work anymore until I manually restart it.
Any ideas to improve this behavior?

ttvt, 2021-09-21:
Hi,
This addon is in development process and not stable enough to publish. 
This add on currently just export today & monthly solar energy. I can help you to export meter if you send me the sample data.
To determine what cause 'unavailable', I need the log right after midnight.

Yop1403, 2021-09-24:
Hi after some restarts the integration has worked well yesterday (see pic). Today the logfile says:

1632435778: Socket error on client <unknown>, disconnecting.
1632435898: New connection from 172.30.32.2 on port 1883.
1632435898: Socket error on client <unknown>, disconnecting.
1632436018: New connection from 172.30.32.2 on port 1883.
1632436018: Socket error on client <unknown>, disconnecting.
1632436138: New connection from 172.30.32.2 on port 1883.
1632436138: Socket error on client <unknown>, disconnecting.
1632436258: New connection from 172.30.32.2 on port 1883.
1632436258: Socket error on client <unknown>, disconnecting.
1632436378: New connection from 172.30.32.2 on port 1883.
1632436378: Socket error on client <unknown>, disconnecting.
1632436498: New connection from 172.30.32.2 on port 1883
[image]
.

Yop1403, 2021-09-24:
What sample data would you need? This is the info available in the Sungrow app - does this help?

[image]
[image]
[image]
[image]
[image]
> Hi,
> This addon is in development process and not stable enough to publish.
> This add on currently just export today & monthly solar energy. I can help you to export meter if you send me the sample data.
> To determine what cause 'unavailable', I need the log right after midnight.

Yop1403, 2021-09-24:
Update: It seems ModbusTCP2MQTT is disconnecting from Mosquitto Broker. I have 1st only restarted ModbusTCP2MQTT. Then I have also restarted Mosquitto Broker. 

**This is the log in ModbusTCP2MQTT:**
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[10:14:15] INFO: Configured'core-mosquitto' mqtt broker.
INFO:root:Create SungrowModbusTcpClient. Model: sungrow-sg5kd
INFO:root:Modbus connected
INFO:root:Configured MQTT Client
INFO:root:Publish Home Assistant Discovery message
WARNING:root:Modbus connection failed
WARNING:root:Failed to scrape inverter, sleeping until next scan
ERROR:pymodbus.factory:Unable to decode response Modbus Error: Unknown response 0
ERROR:pymodbus.transaction:Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/pymodbus/transaction.py", line 190, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/usr/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
WARNING:root:Modbus connection failed
WARNING:root:Failed to scrape inverter, sleeping until next scan
INFO:root:{'daily_power_yield': 11900.0, 'total_power_yield': 107.15, 'internal_temp': 60.5, 'pv1_voltage': 240.2, 'pv1_current': 9.2, 'pv2_voltage': 300.6, 'pv2_current': 9.1, 'total_pv_power': 4968, 'grid_voltage': 245.6, 'inverter_current': 19.6, 'total_active_power': 4814, 'grid_frequency': 50.2, 'export_power': -4319, 'export_power_indicator': 65535, 'power_meter': 494, 'daily_purchased_energy': 2.1, 'daily_energy_consumption': 1600.0, 'total_energy_consumption': 2108.1, 'timestamp': '2021-09-24T10:14:59'}
INFO:root:Published to MQTT

**This is the log I'm getting in Mosquitto Broker after restart:**
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] mosquitto.sh: executing... 
[10:14:47] INFO: SSL is not enabled
[cont-init.d] mosquitto.sh: exited 0.
[cont-init.d] nginx.sh: executing... 
[cont-init.d] nginx.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[10:14:47] INFO: Starting NGINX for authentication handling...
[10:14:47] INFO: Starting mosquitto MQTT broker...
1632442487: mosquitto version 1.6.12 starting
1632442487: |-- *** auth-plug: startup
[10:14:48] INFO: Successfully send discovery information to Home Assistant.
[10:14:48] INFO: Successfully send service information to the Supervisor.
1632442487: Config loaded from /etc/mosquitto/mosquitto.conf.
1632442487: Loading plugin: /usr/share/mosquitto/auth-plug.so
1632442487:  ├── Username/password checking enabled.
1632442487:  ├── TLS-PSK checking enabled.
1632442487:  └── Extended authentication not enabled.
1632442487: Opening ipv4 listen socket on port 1883.
1632442487: Opening ipv6 listen socket on port 1883.
1632442487: Opening websockets listen socket on port 1884.
1632442487: Warning: Mosquitto should not be run as root/administrator.
1632442487: mosquitto version 1.6.12 running
1632442487: New connection from 127.0.0.1 on port 1883.
1632442487: Socket error on client <unknown>, disconnecting.
1632442490: New connection from 172.30.32.1 on port 1883.
1632442490: New client connected from 172.30.32.1 as 2QEvwlYvQkf5kB3eKbTAi6 (p2, c1, k60, u'homeassistant').
1632442498: New connection from 172.30.32.2 on port 1883.
1632442498: Socket error on client <unknown>, disconnecting.

bdog720, 2021-12-16:
bumping this, need help understanding what data I should be giving to Home Assistant Energy Dashboard

Based on Solariot power_meter is house comsumption in W. 

What sample data do you need? Here is my a log from ModbusTCP2MQTT

INFO:root:{'daily_power_yield': 41000.0, 'total_power_yield': 44.97, 'internal_temp': 40.1, 'pv1_voltage': 0.0, 'pv1_current': 0.0, 'pv2_voltage': 0.0, 'pv2_current': 0.0, 'total_pv_power': 0, 'grid_voltage': 245.4, 'inverter_current': 0.0, 'total_active_power': 0, 'grid_frequency': 50.0, 'export_power': 373, 'export_power_indicator': 0, 'power_meter': 373, 'daily_purchased_energy': 3.4, 'daily_energy_consumption': 2800.0, 'total_energy_consumption': 830.0, 'timestamp': '2021-12-16T23:17:04'}

stuchr, 2021-12-18:
The following would be useful

daily_energy_consummption   = Consumed Energy
daily_energy_purchased     (cannot find an option in the energy dashboard).
export_power = Energy Returned to the Grid 

these would be super useful  hopefully they work

closed 2023-06-18 by ttvt