MatterVN/ModbusTCP2MQTT #2: sungrow-sg8kd power consumption

rianbk, 2021-10-04:
Are you able to figure out the power consumption for the sungrow-sg8kd. Generation only seems to really only provide half the data required for anything meaningful

highground88, 2021-10-11:
Try using the SG5KD for your 8 - it's working fine on mine.

ttvt, 2021-12-31:
I've made some adjustment. Try it and let me know

Yop1403, 2022-01-02:
Slowly getting there. The daily energy consumption is expected by HA in kWh - currently some of the values calculated are in Wh which leads to a wrong result in HA (as per HA, I have fed back 5000 kWh into the grid, today 😆 - 5 kWh would be correct).

[image]

Can you divide the following values by 1000 (marked in colour in the graphics below) - then everything should be correct.
* Daily_power_yield
* daily_energy_consumption
* total_energy_consumption
* 
[image]

mingieee, 2022-01-03:
Having the same problem. 

[image]