MatterVN/ModbusTCP2MQTT #6: Entities for "return to grid" and "grid consumption"

Bullischorsch, 2021-10-30:
Hi,

I can't see entities for the items "return to grid" and "grid consumption" though it should be possible to configure, I think.
I have a sungrow sg10rt, also tried to choose model sg5kd, which makes no difference.

regards and thanks for your work

Georg

nvx, 2021-11-07:
> I can't see entities for the items "return to grid" and "grid consumption" though it should be possible to configure, I think.

They come in as attributes on the entity. Unfortunately the energy monitoring stuff in Home Assistant can't make use of the attribute values directly, but it's easy enough to setup a template sensor in your config file to get them as separate entities which you can then use:

```
template:
  - sensor:
      - name: inverter_self_consumption_today
        unit_of_measurement: kWh
        device_class: energy
        state_class: total_increasing
        state: >
          {{ state_attr('sensor.inverter_energy_today', 'daily_energy_consumption')|float / 1000 }}
        availability: >
          {{ states('sensor.inverter_energy_today')!='unavailable' }}
      - name: grid_purchased_today
        unit_of_measurement: kWh
        device_class: energy
        state_class: total_increasing
        state: >
          {{ state_attr('sensor.inverter_energy_today', 'daily_purchased_energy') }}
        availability: >
          {{ states('sensor.inverter_energy_today')!='unavailable' }}
      - name: grid_exported_today
        unit_of_measurement: kWh
        device_class: energy
        state_class: total_increasing
        state: >
          {{ (state_attr('sensor.inverter_energy_today', 'daily_power_yield')|float - state_attr('sensor.inverter_energy_today', 'daily_energy_consumption')|float) / 1000 }}
        availability: >
          {{ states('sensor.inverter_energy_today')!='unavailable' }}
```

ptC7H12, 2021-11-15:
> daily_energy_consumption
> daily_purchased_energy

Where did you find this value? I want to add some values for my, but I don't know how I can access the values...

nvx, 2021-11-16:
> > daily_energy_consumption
> > daily_purchased_energy
> 
> Where did you find this value? I want to add some values for my, but I don't know how I can access the values...

they are attributes on the sensor state, go look at the state in developer tools and it'll show all the attributes

ptC7H12, 2021-11-16:
> they are attributes on the sensor state, go look at the state in developer tools and it'll show all the attributes
As far as I understand is the attribute a Template which you can define yourself. But I don't understand how you bind it to a value from the Modbus. E.g. "daily_purchased_energy" is nowhere definded. 

So my question is: How can I bind the attribute to a Modbus value.

Sorry, but I'm a beginner to HA...

nvx, 2021-11-16:
> So my question is: How can I bind the attribute to a Modbus value.
>
> Sorry, but I'm a beginner to HA...

It's already there by default, at least with my SG5KD inverter they are anyway, to be able to use them in the energy dashboard you need to setup template entities, which is what the config I pasted earlier does. If it's not there for you it might not be supported on your inverter or your inverter might be wired up differently than mine is.

ptC7H12, 2021-11-16:
Now I do understand the purpose of this Codeblock:

[image]

Thank you! You helped me a lot!

ttvt, 2021-12-31:
Added in last commit

closed 2021-12-31 by ttvt

nvx, 2022-01-07:
Note the commit referenced added self solar consumption, and grid import (although there's an issue with this value as per #15), but it didn't add a grid export field which is needed by the HA Energy stuff.

Grid export needs to be calculated as (daily_power_yield-daily_energy_consumption)/1000 kWh (note the latter is the self consumption already added in the commit last week).

It's easy enough to do this calculation in a template as I'd done above, but it'd be nice for it to work out of the box with the HA Energy stuff without having to setup template sensors.