MatterVN/ModbusTCP2MQTT #17: Energy Management Dashboard - Inverter Power Needs to be kWh as a unit of measurement

scottgow, 2022-01-13:
I've been able to implement my inverter within my HA instance and am getting data, however when trying to use the HA energy management dashboard I can only see the following:

- Inverter Daily Energy Consumption
- Inverter Daily Energy Import
- Inverter Energy Monthly
- Inverter Energy Today

I'm expecting to see a Inverter Power which I believe to be a live measurement of the power being produced by my solar inverter. I've been looking around and it appears as this sensor's unit of measurement is a "W" watt... it won't be picked up via the energy management dashboard.

Are you able to review the above and change the unit of measurement from W to kWh and device_class to energy I believe this should resolve the issue.

Happy to discuss further or test as a beta if needed.

Cheers

bdog720, 2022-01-17:
This is my setup in HA

For Grid consumption I'm using "Inverter Daily Energy Import"
For Return to Grid I have my own sensor setup (see code in configuration.yaml below)
For Solar Production I'm using "Inverter Energy Today"

`template:
      sensor:
        - name: grid_exported_today
        unit_of_measurement: kWh
        device_class: energy
        state_class: total_increasing
        state: >
          {{ (state_attr('sensor.inverter_energy_today', 'daily_power_yield')|float - state_attr('sensor.inverter_energy_today', 'daily_energy_consumption')|float) / 1000 }}
        availability: >
          {{ states('sensor.inverter_energy_today')|lower not in ['unavailable','unknown','none'] }}`

scottgow, 2022-01-19:
Thank you, that's great information.

Yop1403, 2022-01-19:
To add some additional info for the developer: After the improvement of the measurement units (W to kWh / kWh to W) in version 0.2.1.1 the following still needs to be improved/corrected (for reference, my inverter is SG5K - this is most likely the same for all Sungrow inverters):

(1) For one of the items the /1000 division is not necessary as the original value is already in kWh. See picture below:
[image]
--> Can you change this so the value of that sensor is displayed correctly?

(2) HA energy monitoring expects a sensor for the **consumption from the grid:**
[image]
Currently only these sensor can be seen by HA Energy Monitoring:
[image]
--> Can you create a sensor that can be seen by HA Energy Monitoring from this value:
[image]

(3) HA energy monitoring also expects a sensor for the **Return to the grid:**
[image]
but currently there is no such value.
--> Is it possible that you create a sensor that is calculated as:
[image] minus 
[image]
In this example the result should be 20.3 kWh (32.1 kWh minus 11.8 kWh = **20.3 kWh**)

Marshalldog, 2022-02-12:
> To add some additional info for the developer: After the improvement of the measurement units (W to kWh / kWh to W) in version 0.2.1.1 the following still needs to be improved/corrected (for reference, my inverter is SG5K - this is most likely the same for all Sungrow inverters):
> 
> **(1) For one of the items the /1000 division is not necessary as the original value is already in kWh.** See picture below: [image] **--> Can you change this so the value of that sensor is displayed correctly?**
> 
> **(2) HA energy monitoring expects a sensor for the **consumption from the grid:**** [image] Currently only these 4 sensors can be seen by HA Energy Monitoring: [image] **--> Can you create a sensor that can be seen by HA Energy Monitoring based on this value: [image]**
> 
> **(3) HA energy monitoring also expects a sensor for the power returned to the grid:** [image] but currently there is no such value. **--> Is it possible that you create a sensor that is calculated as: [image] minus [image] In this example the result should be 20.3 kWh (32.1 kWh minus 11.8 kWh = **20.3 kWh**)**

> To add some additional info for the developer: After the improvement of the measurement units (W to kWh / kWh to W) in version 0.2.1.1 the following still needs to be improved/corrected (for reference, my inverter is SG5K - this is most likely the same for all Sungrow inverters):
> 
> **(1) For one of the items the /1000 division is not necessary as the original value is already in kWh.** See picture below: [image] **--> Can you change this so the value of that sensor is displayed correctly?**
> 
> **(2) HA energy monitoring expects a sensor for the **consumption from the grid:**** [image] Currently only these 4 sensors can be seen by HA Energy Monitoring: [image] **--> Can you create a sensor that can be seen by HA Energy Monitoring based on this value: [image]**
> 
> **(3) HA energy monitoring also expects a sensor for the power returned to the grid:** [image] but currently there is no such value. **--> Is it possible that you create a sensor that is calculated as: [image] minus [image] In this example the result should be 20.3 kWh (32.1 kWh minus 11.8 kWh = **20.3 kWh**)**

I have the same issue

jsword, 2022-11-04:
I only see inverter daily consumption as an option.

I'd like to see the solar production, grid consumption and return to grid. Here is a link to the energy documentation:

https://www.home-assistant.io/docs/energy/

I believe the energy unit is not being set properly.

jsword, 2022-11-04:
It also looks like the dev_class needs to change from "power" to "energy"

https://www.home-assistant.io/integrations/integration/#energy