MatterVN/ModbusTCP2MQTT #28: Sensors Available Through Add-on

lohita, 2022-03-05:
Hi, thanks for creating this great integration - I'm still getting the hang of it. I've only gotten the following sensors which were created when I installed this addon:
[image]

How do I add ones for energy exported, energy imported from the grid, and energy consumed?

ttvt, 2022-06-08:
For energy import/export, you will need a meter

JMEAthletics, 2022-06-20:
Since the most recent update, I've lost the inverter's internal temperature, was this intentional? I was hoping for use of the inverters internal supply voltage and frequency logs too, i don't suppose these will be added at any point?

ttvt, 2022-06-22:
You will have the temperature and voltage sensor if set scan_level to STANDARD or DETAIL.  However you will need to create frequency sensor manually from mqtt data

closed 2022-06-22 by ttvt