MatterVN/ModbusTCP2MQTT #33: Home Assistant OS installation of ModbusTCP2MQTT - s6-overlay-suexec: fatal: can only run as pid 1

Chris-Mow, 2022-05-14:
I have issues all of a sudden communicating with the Sungrow SG5kd. I don't believe i have done anything to break it, but i have started getting the error;
s6-overlay-suexec: fatal: can only run as pid 1
This in the ModbusTCP2MQTT Log, and the overall home assistant system log is showing 
Failed to to call /addons/4ee6ccd4_modbus_inverter/stats - Container addon_4ee6ccd4_modbus_inverter is not running

I am fairly new to home assistant and would love someone to point me in the right direction. I am running the HyperV image of home assistant OS - core-2022.5.2 (i was at 2022.5.4, but tried a restore an older backup of home assistant to 2022.5.2 to see if it was the update which broke it)

I assume i am getting the Container Addon not running error due to the fact that there is the S6 overlay error, but any further diagnosis is a little above my current level of expertise

highground88, 2022-05-16:
I have just installed this version and am getting the same error.

The old version of ModbusTCP2MQTT I had installed for some time now (0.2) works fine, logs show this:
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
```

antonyclarage, 2022-05-20:
same issue.

highground88, 2022-05-22:
> 

I made a fork of this with the minor change apparently necessary but can't import into HA as a repository.... Oh well, I tried.

stniemin, 2022-05-23:
Hoping to see this fixed soon. It should be a trivial change based on the proposed solution that @antonyclarage shared.

amykapernick, 2022-06-06:
Have the same issue, have submitted a PR with the config change so we can use the fix

ttvt, 2022-06-08:
Solved in latest commit

closed 2022-06-08 by ttvt

highground88, 2022-06-12:
Can anyone confirm this is fixed? I still have the same error after reinstall: s6-overlay-suexec: fatal: can only run as pid 1

amykapernick, 2022-06-13:
Still broken for me

amykapernick, 2022-06-13:
It hasn't actually updated any of the changes, the configuration still has the old options

[image]