MatterVN/ModbusTCP2MQTT #40: No daily self consumption or daily grid consumption

helikopter1909, 2022-06-22:
After update to 0.3.3 I don't get any value for daily self consumption or daily grid consumption. It doesn't matter which inverter is selected or which scan level. I only get following data 
[image]

Before update everything working well. My inverter is SG10KTL-M, before update SG8K-D was used. I use home assistant latest revision.

robm73, 2022-06-23:
I have this issue too - the update has completely broken the energy dashboard where it used to work with 0.2.

Could someone show how to configure any sensors so that they can be used with the energy dashboard please?  I am using this with a SG8K-D.  Thanks!

highground88, 2022-06-23:
I think the issue is the states are not 'total_increasing' perhaps?

robm73, 2022-06-24:
Yeah, I think that is the problem too.  I tried to update the template sensor I had created, but with the change in the entities, I can't seem to get the value of the 'sensor.inverter_export_to_grid'.  I was able to do it before when it was an attribute of sensor.inverter_energy_today, but the vale is now the state of the sensor and I don't know how to convert that to a new template sensor with a class of 'total_increasing'.  My old one was like so:
```
  - sensor:
    - name: grid_exported_today
      unit_of_measurement: kWh
      device_class: energy
      state_class: total_increasing
      state: >
        {{ (state_attr('sensor.inverter_energy_today', 'daily_power_yield')|float - state_attr('sensor.inverter_energy_today', 'daily_energy_consumption')|float) / 1000 }}
      availability: >
        {{ states('sensor.inverter_energy_today')|lower not in ['unavailable','unknown','none'] }}

```

What I am trying now (which is not working) is:
```
- sensor:
    - name: grid_export_today
      unit_of_measurement: Wh
      device_class: energy
      state_class: total_increasing
      state: >
        {{ state_attr('sensor.inverter_export_to_grid') }}
      availability: >
        {{ states('sensor.inverter_export_to_grid')|lower not in ['unavailable','unknown','none'] }}

```
This is obviously wrong as it doesn't work, and I can't find any info or examples on how to achieve this.

helikopter1909, 2022-06-24:
Additional comment, maybe it helps to find a solution. I just check the mosquito broker. All necessary data will be received from the broker, incl. daily consumption (self and grit). How can I transfer this data to a HA sensor?
[image]

robm73, 2022-06-27:
So I figured out getting the state value - quite easy in the end.  My new sensor looks like this:
```
- sensor:
      - name: grid_export_today
        unit_of_measurement: kWh
        device_class: energy
        state_class: total_increasing
        state: >
          {{ (states('sensor.inverter_export_to_grid')|float) / 1000}}
        availability: >
          {{ states('sensor.inverter_export_to_grid')|lower not in ['unavailable','unknown','none'] }}

```
So now I can add the grid_export_today sensor to the Energy dashboard, but even though the value is in kWh, it still seems to add to the dashboard the Wh value.  So when I export 5kWh, it looks like I am exporting 50kWh in the Energy Dashboard.
Has anyone else seen this behaviour?  As I said previously, this was working with the v2.2.1 (which I still have running on my old installation of HA), but this update seems to have changed all that...

robm73, 2022-06-28:
Finally figured out what was wrong.  I had the sensor marked as Wh which I did at the start while figuring out how to get the state. When I got the state I also figured out how to divide by 1000, but I forgot to mark it as kWh when I did that so that was throwing out the values.

Also, I was able to figure out how to get the info out of MQTT:

```
mqtt:
  sensor:
    - state_topic: "inverter/SG8KD/registers"
      name: inverter_export_to_grid_mqtt
      unit_of_measurement: "kWh"
      device_class: energy
      state_class: total_increasing
      value_template: "{{ ((value_json.export_to_grid) / 1000) }}"
```

Either of these options will work.

sEdsKi, 2022-07-29:
I too have less information coming into HA after update. I had previously setup custom sensors for current power usage and solar. Can't seem to replicate in this update.

sEdsKi, 2022-07-31:
Is there a way to downgrade to the previous version? Tried restoring from backup but it seems to mix the new sensors with the old sensors. Im not sure what to try next to downgrade,.

robm73, 2022-08-03:
Not sure about rolling back, but I have my sensor back to normal now and working in the Energy dashboard.  My sensors are defined like this:

```
mqtt:
  sensor:
    - state_topic: "inverter/SG8KD/registers"
      name: inverter_export_to_grid_mqtt
      unit_of_measurement: "kWh"
      device_class: energy
      state_class: total_increasing
      value_template: "{{ (value_json.daily_export_energy) }}"
    - state_topic: "inverter/SG8KD/registers"
      name: daily_import_energy_mqtt
      unit_of_measurement: "kWh"
      device_class: energy
      state_class: total_increasing
      value_template: "{{ (value_json.daily_import_energy) }}"

```

You could use this method to extract any of the MQTT returned data to make your own sensor.

sEdsKi, 2022-08-03:
Thanks, robm73.

I opened a further issue - as the iSolarCloud app was weird after updating. (disconnecting and sending incorrect data to the cloud) If I can get that issue sorted I will definitely give the MQTT sensors a go.

helikopter1909, 2022-08-03:
I found a solution for my configuration, maybe it will help others too

sensor:
 - platform: mqtt
   name: daily export energy
   unique_id: daily_export_energy
   state_topic: inverter/SG8KD/registers
   value_template: '{{ value_json.daily_export_energy }}'
   unit_of_measurement: kWh
   icon: mdi:solar-power      
   device_class: energy
   state_class: total_increasing
 - platform: mqtt
   name: daily import energy
   unique_id: daily_import_energy
   state_topic: inverter/SG8KD/registers
   value_template: '{{ value_json.daily_import_energy }}'
   unit_of_measurement: kWh
   icon: mdi:solar-power 
   device_class: energy
   state_class: total_increasing

closed 2022-08-03 by helikopter1909