MatterVN/ModbusTCP2MQTT #43: Can multiple inverters be added?

zztopgtb, 2022-07-26:
I have 2x Sungrow SG5K-D inverters. This add-on appears to only pull data from the specified host. Can 2x hosts be specified? Thanks

shansted, 2022-08-11:
I also have this setup, with one inverter having the smart meter, the other not. If I only read one inverter, the feed in will be correct, but solar production will not. I was using multiple instances of SolarIOT, but an addon that does not require manual mqtt configuration and container installation would be nice.

zztopgtb, 2022-08-15:
How did you setup SolarIOT? Theres no add-on that I can see.

shansted, 2022-08-18:
Yeah, no HA addon. I was using portainer to install containers manually. They've since dropped portainer as an addon, which brought me here since a reinstall.

Its probably not particularly common to have this setup. I've been meaning to get around to understanding how HA addons work, so this will be a good start project. I'll first try to extend for our use case (2 of the same inverters), and then extend out to support multiple.

Probably not a big deal for someone who knows what they are doing, but my programming skills here are limited.

ttvt, 2022-11-05:
I don't have plan to support multiple inverters.  Will think about it later.

closed 2022-11-05 by ttvt