MatterVN/ModbusTCP2MQTT #44: Writing data to external app

sEdsKi, 2022-08-02:
Since updating, the app I use to view solar inverter data directly is behaving strangely.

It's either disconnected from the internet and not uploading the data or it seems to be writing data to the inverter.

Neither happened in the previous version.

[image]

[image]

highground88, 2022-08-02:
[image]

Mine is running on the latest version with no apparent reading issues. But a week or two ago, I noticed that my inverter was showing as offline on iSolarCloud for a few days. Unsure if it was coincidental, but this was similar timings to the update. I removed the add-on and reinstalled (removed any remnants) and everything appears working again.

sEdsKi, 2022-08-02:
What equipment do you have?

I have the sg8k-d with wifi dongle

highground88, 2022-08-02:
Exactly the same mate.

sEdsKi, 2022-08-03:
> highground88

Could you share your config settings?
I have just removed the addon and will now reinstall it.

sEdsKi, 2022-08-05:
I've reset the inverter hardware and reinstalled the add-on which seems to have reset everything to normal. Data looks normal for yesterday. Today I see the isolarcloud app stopped receiving data at 12.20pm

[image]

[image]
 
The net light on the wifi dongle is off.

Data is still being received by Home Assistant

[image]

Can't say it's this add-on causing that problem... But I would guess it is.

sEdsKi, 2022-08-07:
Have pulled out the Wifi dongle and replaced it. Data now flowing into iSolarCloud App and HA.
I will update if anything changes.

helikopter1909, 2022-09-22:
I have same problem after update 2022.9.6 and to new config for mtqq source. data in ha available and not in isolarcloud App. Has anybody a solution? How do config your mtqq sensor in configuration.yaml?

helikopter1909, 2022-09-24:
The described problem is a bug in the isolarcloud app and not a problem with the mod.

michael-letcher, 2022-12-02:
Having the same issue with my inverter. After a period of time, the app stops getting data and only after stopping the HA Add-on does it start again.

jsword, 2023-01-15:
I've seen the same think after power/network outages. iSolar cloud stops but HA still receives data. Pulling the dongle fixes it.

closed 2023-06-18 by ttvt