MatterVN/ModbusTCP2MQTT #45: FATAL: No internal MQTT Broker found

arfrater, 2022-08-07:
Hi 
Still getting this error on Current version: 0.3.3.1 
curl: (7) Couldn't connect to server
[13:58:12] ERROR: Something went wrong contacting the API
[13:58:13] FATAL: No internal MQTT Broker found. Please install Mosquitto broker.

elektrinis, 2022-08-19:
I have a similar problem:

s6-rc: info: service legacy-services successfully started
[19:34:40] FATAL: No internal MQTT Broker found. Please install Mosquitto broker.
s6-rc: info: service legacy-services: stopping

Mosquitto broker v6.1.2. is installed and running.

jullit31, 2022-08-25:
@arfrater  I assume you have Mosquitto broker installed via the HA add-on store?

arfrater, 2022-08-25:
> @arfrater I assume you have Mosquitto broker installed via the HA add-on store?

Yes I have

elektrinis, 2022-08-31:
In my case it started working once I have manually restarted MQTT plugin.

arfrater, 2022-08-31:
> In my case it started working once I have manually restarted MQTT plugin.

That never worked for me tried to reinstall and still error

ttvt, 2022-11-05:
Please help me to test this case and let me know it can help you:
Restart the Broker; after a while restart the Modbus addon.

closed 2023-06-18 by ttvt