MatterVN/ModbusTCP2MQTT #47: Only 9 sensors get auto-discovered

jullit31, 2022-08-14:
I just installed the add-on and connected to an SH6.0RT inverter with smart meter, `scan_level` is set to `DETAIL` . However, MQTT auto-discovered only 9 sensors:
[image]

The full MQTT message contains a lot more sensors though:
```
QoS: 0
Payload: device_type_code: SH6.0RT
run_state: 'ON'
protocol_number: 1094856704
protocol_version: 16781568
arm_software_version: 16722
dsp_software_version: 19780
nominal_active_power: 6
output_type: 3P4L
daily_power_yields: 30.9
total_power_yields: 24279
internal_temperature: 43.1
mppt_1_voltage: 247
mppt_1_current: 0
mppt_2_voltage: 509.4
mppt_2_current: 1.3
total_dc_power: 615
phase_a_voltage: 232.2
phase_b_voltage: 232.2
phase_c_voltage: 233.8
total_reactive_power: -5
power_factor: 1
grid_frequency: 49.9
active_power_regulation_setpoint: 0
meter_power: 0
load_power: 400
daily_running_time: 0
pv_power_of_today: 0
daily_pv_energy_yields: 24.9
monthly_pv_energy_yields: 0
yearly_pv_energy_yields: 0
direct_power_consumption_today_pv: 0
direct_power_consumption_pv: 4.5
direct_power_consumption_monthly_pv: 0
direct_power_consumption_yearly_pv: 0
export_power_from_pv_today: 0
export_power_from_pv: 15.4
export_power_from_pv_monthly: 0
export_power_from_pv_yearly: 0
battery_charge_power_from_pv_today: 0
battery_charge_power_from_pv: 5
battery_charge_power_from_pv_monthly: 0
battery_charge_power_from_pv_yearly: 0
system_state: Run
running_state: 11
power_generated_from_pv: 1
battery_charging: 1
battery_discharging: 0
load_active: 1
feed_into_grid: 0
import_from_grid: 0
power_generated_from_load: 0
daily_pv_generation: 35.5
total_pv_generation: 2557.9
daily_pv_export: 20.9
total_pv_export: 1824.4
load_power_hybrid: 396
export_power_hybrid: 4
daily_battery_charge_from_pv: 7
total_battery_charge_from_pv: 413.7
co2_reduction: 1790.5
daily_direct_energy_consumption: 7.6
total_direct_energy_consumption: 319.8
battery_voltage: 320.2
battery_current: 0.5
battery_power: 149
battery_level: 96.8
battery_state_of_healthy: 100
battery_temperature: 25
daily_battery_discharge_energy: 3.2
total_battery_discharge_energy: 367.5
self_consumption_of_day: 26.6
grid_state: 0
phase_a_current: 0.7
phase_b_current: 0.7
phase_c_current: 0.7
total_active_power: 400
daily_import_energy: 0
total_import_energy: 37.8
daily_charge_energy: 7
total_charge_energy: 412.2
drm_state: 255
daily_export_energy: 20.9
total_export_energy: 1834.4
start_stop: Start
power_limitation_switch: Enable
power_limitation_setting: 100
power_factor_setting: 0
reactive_power_adjustment_mode: 'Off'
reactive_power_percentage_setting: 0
power_limitation_adjustment: 0
reactive_power_adjustment: 0
export_to_grid: 0
timestamp: '2022-8-14 18:26:59'
```
Is there anywhere I can set which sensors are shown?

julianbadell, 2022-08-24:
same issue here

cartman10, 2022-08-29:
having the issue as well

elektrinis, 2022-09-04:
only 6 sensors with SH10RT...

Falkoh150, 2022-09-16:
Same here, only 9 Sensors.

elektrinis, 2022-10-22:
I have added most of main parameters in the new pull request. Hopefully the owner will merge it.

DIYGuy01, 2022-11-02:
agree, from my SG12RT I see just 2 :( . Please merge @elektrinis change
[image]

DIYGuy01, 2022-11-03:
@elektrinis I have SG12RT. After merge I have got +4 sensors but still not complete list of scraped modbus registers. Debug shows many registers return the state but some not appear as entities?
```
2022-11-03 22:25:47 INFO     Inverter: Successfully scraped in 1.747578 secs
2022-11-03 22:25:47 DEBUG    MQTT: Publishing: inverter/sg12rt/registers : {"device_type_code": "sg12rt", "run_state": "ON", "nominal_active_power": 12.0, "output_type": "3P4L", "daily_power_yields": 5.5, "total_power_yields": 93, "internal_temperature": 25.2, "mppt_1_voltage": 0.0, "mppt_1_current": 0.0, "total_dc_power": 0, "phase_a_voltage": 0.0, "phase_b_voltage": 0.0, "phase_c_voltage": 0.0, "total_reactive_power": 0, "power_factor": 0.0, "grid_frequency": 0.0, "daily_running_time": 516, "start_stop": "Start", "power_limitation_switch": "Enable", "power_limitation_setting": 110.0, "power_factor_setting": 0.0, "reactive_power_adjustment_mode": "Enable Q(U)", "reactive_power_percentage_setting": 0.0, "power_limitation_adjustment": 13.2, "reactive_power_adjustment": 0.0, "export_to_grid": 0, "import_from_grid": 0, "timestamp": "2022-11-3 22:25:45"}
2022-11-03 22:25:47 INFO     MQTT: Published
2022-11-03 22:25:47 DEBUG    Processing Time: 1.75 secs
```
[image]

ttvt, 2022-11-05:
I don't have time to create sensors from scraped data. Any help will be appreciated.

Falkoh150, 2022-11-05:
After the update, all important data is now displayed for me. Many Thanks.

jullit31, 2022-11-05:
I also just got an additional 4 sensors:
[image]

elektrinis, 2022-11-05:
Sorry guys I can't add/test additional entities for SG series, as mine is SH... Feel free to experiment by modifying the file using other entities as example.

andi-blafasl, 2022-11-21:
How do I get the additional Values? I just added a Battery to my SH6.0RT but nothing changed in the Addon.
[image]
I see everything in the MQTT messages but they don't get exposed to Home Assistant.

andi-blafasl, 2022-11-23:
@elektrinis how can I get your updated version into my home assistant installation? Do I need to modify any files? The addon installed is version 0.3.8
@ttvt is 0.3.8 the latest version with the additions from elekrinis? If yes, why am I missing the Sensors? Any way to rediscover the inverter or something?

elektrinis, 2022-11-28:
> @elektrinis how can I get your updated version into my home assistant installation? Do I need to modify any files? 

In the end I have just installed it as custom addon and experimented with different options. Just copy the files to /addons/ folder and then make the changes you want. Note this folder is only accessible via samba share, and then it appears as available in the store. Google around to get more info how it's done.

However I don't really understand how this works, as some of my entities, like "Battery level" just became unavailable, seemingly without a reason. They are still available in mqtt messages.
Also, when added to the file, not all parameters appear as entities. Not sure how to debug this.

Also please note that sungrow SH series can't be trusted, as a lot of data coming from it is fake. For example, ingoing and outgoing power is always equal, however there should be some losses (especially evident due to it running +15C from ambient).
Also this, multiplying current*voltage should be 1353W, but is reported as 640W.
[image]

andi-blafasl, 2022-11-29:
@elektrinis thank you for the detailed explanation.
Tried to add some sensors for the Batterie by configuring them via MQTT in config.yaml. But I found out that the "Daily Generation" is no longer usable for me. It includes generation from batterie discharging and that doesn't make any sense when using it in the energy dashboard.
I just switched to a more simplistic approach via direkt modbus integration. https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant
Here you get all the registers exposed as entities and also you can control your inverter from within Home Assistant. It worked for me out of the box.

elektrinis, 2022-11-29:
This looks super cool, will certainly test it. Thank you.