MatterVN/ModbusTCP2MQTT #56: Grid consumption and Return to grid for energy dashbord missing

cartman10, 2022-12-26:
Only Inverter Daily Generation can be selected in the dashboard confiration page. 
Smart_meter is set to true
[image]
Sungrow SG10RT with WiNet-S_001_247
Do I need to configue something in addition to see the missing statitsics?

BigChaps, 2023-02-03:
I am trying to work this out (I have a SG08RS) but I am an Infrastructure guy first, coding second.  

From what I understand, it is because those sensors don't have a state or device class so it needs to be defined in the configuration.yaml. Any suggestions would be appreciated but if I work it out, I will post it here.

kirangp, 2023-12-12:
@BigChaps Did you ever figure it out?