MatterVN/ModbusTCP2MQTT #59: Modbus Error: [Connection] Failed to connect[ModbusTcpClient

scraja69, 2023-03-29:
Hi 
I managed to get data through Winet-S dongle one month back. Now it is not giving any data... 
Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.125:502)]')

Do I have to coneect to native LAN port.?  I prefer to use local ethernet. Model :SG15RT

023-03-29 10:29:15 WARNING  No data returned for read, 5600:100
No data returned for read, 5600:100
2023-03-29 10:29:15 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.125:502)]')
2023-03-29 10:29:15 DEBUG    Scraping: read, 6099:100
2023-03-29 10:29:15 DEBUG    load_registers: read, 6099:100
2023-03-29 10:29:15 ERROR    Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
2023-03-29 10:29:15 WARNING  No data returned for read, 6099:100
No data returned for read, 6099:100
2023-03-29 10:29:15 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.125:502)]')
2023-03-29 10:29:15 DEBUG    Scraping: read, 6199:100
2023-03-29 10:29:15 DEBUG    load_registers: read, 6199:100
2023-03-29 10:29:15 ERROR    Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
2023-03-29 10:29:15 WARNING  No data returned for read, 6199:100
No data returned for read, 6199:100
2023-03-29 10:29:15 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.125:502)]')
2023-03-29 10:29:15 DEBUG    Scraping: read, 6299:100
2023-03-29 10:29:15 DEBUG    load_registers: read, 6299:100
2023-03-29 10:29:15 ERROR    Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
2023-03-29 10:29:15 WARNING  No data returned for read, 6299:100
No data returned for read, 6299:100
2023-03-29 10:29:15 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.125:502)]')
2023-03-29 10:29:15 DEBUG    Scraping: read, 6399:100
2023-03-29 10:29:15 DEBUG    load_registers: read, 6399:100
2023-03-29 10:29:15 ERROR    Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
2023-03-29 10:29:15 WARNING  No data returned for read, 6399:100
No data returned for read, 6399:100
2023-03-29 10:29:15 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.125:502)]')
2023-03-29 10:29:15 DEBUG    Scraping: read, 6499:100
2023-03-29 10:29:15 DEBUG    load_registers: read, 6499:100
2023-03-29 10:29:15 ERROR    Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
2023-03-29 10:29:15 WARNING  No data returned for read, 6499:100
No data returned for read, 6499:100
2023-03-29 10:29:15 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.125:502)]')
2023-03-29 10:29:15 DEBUG    Scraping: read, 6599:100
2023-03-29 10:29:15 DEBUG    load_registers: read, 6599:100
2023-03-29 10:29:15 ERROR    Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
2023-03-29 10:29:15 WARNING  No data returned for read, 6599:100
No data returned for read, 6599:100
2023-03-29 10:29:15 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.125:502)]')
2023-03-29 10:29:15 DEBUG    Scraping: read, 6699:100
2023-03-29 10:29:15 DEBUG    load_registers: read, 6699:100
2023-03-29 10:29:15 ERROR    Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
2023-03-29 10:29:15 WARNING  No data returned for read, 6699:100
No data returned for read, 6699:100
2023-03-29 10:29:15 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.125:502)]')
2023-03-29 10:29:15 DEBUG    Scraping: read, 7012:25
2023-03-29 10:29:15 DEBUG    load_registers: read, 7012:25
2023-03-29 10:29:15 ERROR    Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
2023-03-29 10:29:15 WARNING  No data returned for read, 7012:25
No data returned for read, 7012:25
2023-03-29 10:29:15 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.125:502)]')
2023-03-29 10:29:15 DEBUG    Scraping: read, 12999:125
2023-03-29 10:29:15 DEBUG    load_registers: read, 12999:125
2023-03-29 10:29:15 ERROR    Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
2023-03-29 10:29:15 WARNING  No data returned for read, 12999:125
No data returned for read, 12999:125
2023-03-29 10:29:15 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.125:502)]')
2023-03-29 10:29:15 DEBUG    Scraping: hold, 4999:10
2023-03-29 10:29:15 DEBUG    load_registers: hold, 4999:10
2023-03-29 10:29:15 ERROR    Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
2023-03-29 10:29:15 WARNING  No data returned for hold, 4999:10
No data returned for hold, 4999:10
2023-03-29 10:29:15 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.125:502)]')
2023-03-29 10:29:15 DEBUG    Scraping: hold, 5009:10
2023-03-29 10:29:15 DEBUG    load_registers: hold, 5009:10
2023-03-29 10:29:15 ERROR    Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
2023-03-29 10:29:15 WARNING  No data returned for hold, 5009:10
No data returned for hold, 5009:10
2023-03-29 10:29:15 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.125:502)]')
2023-03-29 10:29:15 DEBUG    Scraping: hold, 5034:10
2023-03-29 10:29:15 DEBUG    load_registers: hold, 5034:10
2023-03-29 10:29:15 ERROR    Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
2023-03-29 10:29:15 WARNING  No data returned for hold, 5034:10
No data returned for hold, 5034:10
2023-03-29 10:29:15 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.125:502)]')
2023-03-29 10:29:15 DEBUG    Scraping: hold, 13000:100
2023-03-29 10:29:15 DEBUG    load_registers: hold, 13000:100
2023-03-29 10:29:15 ERROR    Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
Connection to (192.168.1.125, 502) failed: [Errno 111] Connection refused
2023-03-29 10:29:15 WARNING  No data returned for hold, 13000:100
No data returned for hold, 13000:100
2023-03-29 10:29:15 DEBUG    Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.125:502)]')
2023-03-29 10:29:15 INFO     Disconnecting: ModbusTcpClient(192.168.1.125:502)
2023-03-29 10:29:15 INFO     Disconnecting: None
2023-03-29 10:29:15 WARNING  Data collection failed, skipped exporting data. Retying in 30 secs
Data collection failed, skipped exporting data. Retying in 30 secs
2023-03-29 10:29:15 DEBUG    Processing Time: 0.15 secs
2023-03-29 10:29:15 INFO     Next scrape in 29 secs

piyushbhatia10, 2023-04-19:
I am facing the same issue.

scraja69, 2023-04-19:
Hi,
I have contacted Sungrow and able to get modbus connection as given below...
[image]

[The Procedure for Setting up Winet Modbus Connection.pdf](https://github.com/MatterVN/ModbusTCP2MQTT/files/11268166/The.Procedure.for.Setting.up.Winet.Modbus.Connection.pdf)
[TD_202006_RS485 Communication Checklist_V1.0.pdf](https://github.com/MatterVN/ModbusTCP2MQTT/files/11268179/TD_202006_RS485.Communication.Checklist_V1.0.pdf)