MatterVN/ModbusTCP2MQTT #62: This addon is no longer appearing in HA addon store.

Shammy01011101, 2023-05-30:
I have this addon and its working great.  Had to start troubleshooting some error logs lately and noticed that this addon is no longer in the addon store.  I can only see three addons under your repository.  

They are:
Hass.io Community Store
Home Assistant Google Drive Backup
Zigbee2MQTT

I have tried re adding the repository however this has not rectified the issue. 
[image]

Shammy01011101, 2023-06-11:
Cmd('git') failed due to: exit code(128) cmdline: git clone -v --recursive --depth=1 --shallow-submodules -- https://github.com/MatterVN/HassioAddon /data/addons/git/eeb054de stderr: 'Cloning into '/data/addons/git/eeb054de'... POST git-upload-pack (175 bytes) POST git-upload-pack (244 bytes) Submodule 'ModbusTCP2MQTT' (https://github.com/MatterVN/ModbusTCP2MQTT) registered for path 'ModbusTCP2MQTT' Submodule 'TTLock2MQTT' (https://github.com/MatterVN/TTLock2MQTT) registered for path 'TTLock2MQTT' Cloning into '/data/addons/git/eeb054de/ModbusTCP2MQTT'... Cloning into '/data/addons/git/eeb054de/TTLock2MQTT'... fatal: remote error: upload-pack: not our ref a48590a7a7b16dc003336e4d6a7d7671c372979a fatal: Fetched in submodule path 'ModbusTCP2MQTT', but it did not contain a48590a7a7b16dc003336e4d6a7d7671c372979a. Direct fetching of that commit failed. '

ttvt, 2023-06-18:
Fixed.

closed 2023-06-18 by ttvt