MatterVN/ModbusTCP2MQTT #63: The service starts and stops

saeedesmaili78, 2023-06-18:
Hello,

The ModbusTCP2MQTT starts and stops.

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[00:06:25] INFO: Configured'core-mosquitto' mqtt broker.
[00:06:25] INFO: Generated config file.
2023-06-19 00:06:26 INFO     Starting SunGather 0.3.8
2023-06-19 00:06:26 INFO     Loaded config: config.sg
2023-06-19 00:06:28 INFO     Loaded registers: //registers-sungrow.yaml
2023-06-19 00:06:28 INFO     Registers file version: 0.2.0
Traceback (most recent call last):
  File "/sungather.py", line 596, in <module>
    main()
  File "/sungather.py", line 506, in main
    logger.handlers[0].setLevel(config_inverter['log_console'])
  File "/usr/lib/python3.11/logging/__init__.py", line 940, in setLevel
    self.level = _checkLevel(level)
                 ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/logging/__init__.py", line 207, in _checkLevel
    raise ValueError("Unknown level: %r" % level)
ValueError: Unknown level: 'TRUE'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

saeedesmaili78, 2023-07-14:
Hi,
Are you running the home assistant on a single-board computer like Raspberry pi or Asus Tinker board? 
I faced this issue on a tinker board. On a PC it works without issue.

DariusNeciunas, 2023-07-14:
Hi,
HA running on raspberrPI.