MatterVN/ModbusTCP2MQTT #65: Cannot install addon

marten-lucas, 2023-07-19:
When I click "install" on the page in the addon store, I recieve the following message:

Addon could not be installed
`
The command '/bin/ash -o pipefail -c pip install --upgrade pycryptodomex==3.11.0 --no-cache-dir -r requirements.txt' returned a non-zero code: 1`

ergi-la, 2023-07-25:
Same problem here.
It might be the requirement readsettings>=3.4.5 it is not possible to install this with pip in actual HA Version.
The package readsettings ist outdated i think.
I don´t know how to fix this problem ...

hansen1506, 2023-07-25:
Same here.

Home Assistant 2023.7.3
Supervisor 2023.07.1
Operating System 10.3
Frontend 20230705.1 - latest

silviopen, 2023-08-03:
I also join the list of users with this problem.
Has anyone solved it?

wombat1971, 2023-08-03:
Same issue. Have to start a new build of HA but cannot install.

Madcat71, 2023-08-21:
Same issue

alpham1ke, 2023-08-27:
+1 on this issue, with setup below:
Home Assistant 2023.8.4
Supervisor 2023.08.1
Operating System 10.5
Frontend 20230802.1 - latest

and same error reported about a year ago, which self resolved (?) - a "thankyou" noted but no other responses in the thread?

https://github.com/MatterVN/ModbusTCP2MQTT/issues/38

wileecotye, 2023-09-03:
same issue here -i manually installed the file and checked the version -and got a error installing readsetting -warning about the setup license file parameter being deprecated -there is a link but its beyond my skill set 
(https://setuptools.pypa.io/en/latest/userguide/declarative_config.html)

m0ister, 2023-09-05:
Same issue

MrTuuist3r, 2023-09-10:
Same issue :-(

HunZ83, 2023-09-12:
i´m also not able to install the addon.
Home Assistant 2023.9.1
Supervisor 2023.08.3
Operating System 10.5
Frontend 20230802.0 - latest

insanetypo, 2023-10-03:
Same problem for me with latest version installed (10.5)

dimafemabo, 2023-10-14:
same issue here :-(
Home Assistant 2023.3.3 Supervisor 2023.10.0
HA in Docker
Anyone a step further or solved the issue?
Cheers

dimafemabo, 2023-10-15:
> When I click "install" on the page in the addon store, I recieve the following message:
> 
> Addon could not be installed ` The command '/bin/ash -o pipefail -c pip install --upgrade pycryptodomex==3.11.0 --no-cache-dir -r requirements.txt' returned a non-zero code: 1`

As I have the same issue, see my comment above, I realized the comment uses "/bin/ash -o p" shouldn't that be *bash* instead of ash???
Could this be corrected somwhere in the script??
any ideas?
Cheers and have a great weekend!

shaunsull, 2023-10-23:
Same issue 
```
Home Assistant 2023.10.5
Supervisor 2023.10.0
Operating System 11.0
Frontend 20231005.0 - latest
```

shaunsull, 2023-11-05:
In the docker file, should the version of [pycryptodomex](https://pypi.org/project/pycryptodomex/) be 3.19.0?

`RUN pip install --upgrade pycryptodomex==3.11.0 --no-cache-dir -r requirements.txt `

Sorry I am not good enough at GitHub to do a pull request.

ashmieke, 2023-11-09:
I managed to fix by updating requirements.txt first line to PyYAML==5.3.1

shaunsull, 2023-11-11:
> I managed to fix by updating requirements.txt first line to PyYAML==5.3.1

That's great @ashmieke 

For someone who is not good at GitHub, how would I go about installing into HA with your code fix still in a pull request? In other words, what are the steps we need to do before it get's code commitment?

plumpy80, 2023-11-14:
Of course, the same issue.... How can I use the soulutions of @ashmieke  ?

plumpy80, 2023-11-24:
Unfortunately, it still doesn't work...   😄 or 😢

wombat1971, 2023-11-24:
Noticed another update to the jason file overnight (for Australia). It has installed.. no error. Now trying to configure.

shaunsull, 2023-11-25:
Confirmed working again. Many thanks.