MatterVN/ModbusTCP2MQTT #68: Help needed setting this up

KLucky13, 2023-12-05:
Hello,

I have 2 inverters:
1 x SG2.0RS-S
1 x SH4.0RS

I've been using this integration until today but it has been giving me nothing but problems since it goes through the api of isolarcloud: https://github.com/MickMake/GoSunGrow/

So I want to use an integration that bypasses the isolarcloud api/connection.

However I'm not sure how to proceed with this integration. Both my inverters have a Wifi-S dongle/module but I could also connect them with Ethernet to the dongle. If I unplug the dongle all together, I also have an ethernet connector on the converter.

What would be the way forward? Can I just leave them as is (connected to the wifi)? If yes, I'm not sure where I can find the ip address of these dongles; I've looked everywhere in the Isolarcloud app / Webpage but it seems it's not something I can find.

Looking at my router, I've 100 devices or more, so can't tell which one is which.

Any help setting this up would be very much appreciated!

dimafemabo, 2023-12-05:
Hi,
I loved this integration and will be patient until an update comes out, however, if you find a good modbus integration that shows also the plethora of information, I am happy to hear about it.
To answer your question, try to look out for an entry in your router (mine is FB) where the name says "espressif".
Also I do have 2 inverters and as they both have been connected to my router at the same time, their IP address is just ...91 and ...92.
When I connected the modbus via the extra LAN cable the router assigned the next free IP address to the lan connection.
Also here the connection was called "esspressif".
Good luck and keep us posted.
Cheers

KLucky13, 2023-12-06:
> As I see, it has no options for a second inverter :(

That's too bad, I have an inverter for every side of the roof since I have more solar panels on one side of the roof compared to the other one. So I would need to be able to add both off them, otherwise there is no point implementing this solution :-/

ventudan, 2024-05-09:
Good evening everyone. I'm not familiar with Home Assistant...so my question will seem trivial...
but how do I add a card once the Add-on has been installed??

DarthKegRaider, 2024-07-24:
> Good evening everyone. I'm not familiar with Home Assistant...so my question will seem trivial... but how do I add a card once the Add-on has been installed??

@ventudan:  
Open HomeAssistant.
Click "Settings" on the left navigation panel.
Click "Devices and Services" in the middle panel.
Click "MQTT" application.
Click the xx "devices".  Where xx is a number of detected devices.
Click on your inverter model.  Mine was "SG8K-D".
Now you can click the "Add to Dashboard" and this card will be added whatever panel you select.

I hope this is what you wanted.