MatterVN/ModbusTCP2MQTT #72: Unable to install the addon

kirangp, 2023-12-12:
I had to reinstall the addon and now whenever I try to install, I always get the below error. 

[image]

Thanks

mife1005, 2023-12-14:
I have the same problem. Hoping for a solution!

anthony-spruyt, 2023-12-14:
Pretty much all the sungrow addons have this problem atm. Tried pinning some different versions but no luck.

Ralphy140, 2023-12-19:
Same issue for me

SoWizard, 2023-12-27:
Yup ditto - same issue here.

loucksg, 2023-12-27:
k, so it does not look like the:
pycryptodomex==3.11.0
Is in the main requirements, however I am cautious this is inside the Docker container... just need to work out how to get inside that and the wrapping here, as a blunt hammer may try:
pycryptodomex~=3.11.0
for requirements.txt, that should install anything compatible with 3.11 code stream...

loucksg, 2023-12-27:
found it in Dockerfile:
COPY requirements.txt ./
RUN apk add --no-cache python3-dev py3-pip g++
RUN pip install --upgrade pycryptodomex==3.11.0 --no-cache-dir -r requirements.txt

Changing to:
COPY requirements.txt ./
RUN apk add --no-cache python3-dev py3-pip g++
RUN pip install --upgrade pycryptodomex~=3.11.0 --no-cache-dir -r requirements.txt

... let's see what goes boom.

loucksg, 2023-12-27:
pull request created, not sure if someone needs to approve here, will dig a little further around repo

ltw001, 2023-12-27:
same here. Also freshly installed Home Assistant OS running in a VM.

SoWizard, 2023-12-30:
@ltw001 Link seems broken?

kirangp, 2024-01-04:
@loucksg Is there any way I can install this addon with your pull request in HA? Looks like the pull request has not been approved.

benborra, 2024-01-12:
> @loucksg Is there any way I can install this addon with your pull request in HA? Looks like the pull request has not been approved/merged.

I'm guessing we could fork https://github.com/MatterVN/HassioAddon and this repo and fix it on that fork?
I've tried that but I'm rather new to Home Assistant add-ons and can't seem to get the reference from the HassioAddon repo to the submodule to work.

kilador, 2024-01-15:
Same problem here. I try to install the addon following the instructions but still does not work.

is it possible to upload with the fix ?

sharkpunch5, 2024-02-05:
+1 same issue here when attempting to install

SL666, 2024-02-29:
> @ltw001 Link seems broken?

I think it's this - 
https://gist.github.com/Paraphraser/cad3b0aa6428c58ee87bc835ac12ed37#part-2--using-patched-images-in-homeassistant

but seems to be referencing gosungrow not modbustcp2mqtt

RafAustralia, 2024-05-03:
same here totally fresh here - how to fix this? - please help

benborra, 2024-05-03:
> same here totally fresh here - how to fix this? - please help

Hi @RafAustralia, this was fixed in a fork over at https://github.com/DisreputableCode/HassioAddons
Been using it from r about a month or 2 without any problems.

RafAustralia, 2024-05-03:
Hey Benborra.

Firstly huge thank you for replying.
It means a lot.

Secondly please note I am so new to home assistant that I would barely know how to apply this fix. I just read through It and it has the fixes I saw people mention online like the docker file etc etc.
But it is no longer pressing an install button  stuff. It needs knowledge of code.
Im getting better by the minute but my code is minimal. Been doing it for 3 days. Lol. Sorry.

Hencewhy its a bit of babysitting stuff

Raf
Sent from Nam Concept’s iPhone

RafAustralia, 2024-05-04:
Hey!

It actually worked… I was able to install it, but by the time I got there some of my sensors were so messed up that I decided to start over… but I just wanted to say huge thank you for being so kind and showing me the fix for it. 😊

Thank you.

With warmest regards,
[NAM concept]<https://htmlsig.com/t/0000001BT6QSA>
Rafal Tomaszewski / Director
***@***.******@***.***> / +61404089858
NAM concept
Office: +61 439 348 281
116 King Arthur Tce, Tennyson, QLD 4105
PO BOX 2228, Graceville East, QLD 4075
www.nam-concept.com.au<http://www.nam-concept.com.au/>
[htmlsig.com]<https://htmlsig.com/t/0000001BV8W1D>

This e-mail message may contain confidential or legally privileged information and is intended only for the use of the intended recipient(s). Any unauthorized disclosure, dissemination, distribution, copying or the taking of any action in reliance on the information herein is prohibited. E-mails are not secure and cannot be guaranteed to be error free as they can be intercepted, amended, or contain viruses. Anyone who communicates with us by e-mail is deemed to have accepted these risks. NAM Concept is not responsible for errors or omissions in this message and denies any responsibility for any damage arising from the use of e-mail. Any opinion and other statement contained in this message and any attachment are solely those of the author and do not necessarily represent those of the company.

benborra, 2024-05-04:
No worries, was one of the most complex things I've done so far in home assistant as well.
I unfortunately don't recall which guide I followed but I also had some issues fetching the data from my sungrow converter because I was polling to frequently. Couldn't even reach it directly by IP anymore myself.

I also had to add the MQTT to my Integrations.
[image]

And there I needed to add my parameters.

[image]

[image]