MatterVN/ModbusTCP2MQTT #74: Failed to install add-on

Korsberg, 2024-01-12:
Failed to install add-on
The command '/bin/ash -o pipefail -c pip install --upgrade pycryptodomex==3.11.0 --no-cache-dir -r requirements.txt' returned a non-zero code: 1

benborra, 2024-01-12:
Hey @Korsberg, this has been mentioned in https://github.com/MatterVN/ModbusTCP2MQTT/issues/72 and should be fixed with https://github.com/MatterVN/ModbusTCP2MQTT/pull/73

DIYGuy01, 2024-02-10:
> Hey @Korsberg, this has been mentioned in #72 and should be fixed with #73

does it mean death of this addon, project closure ?

benborra, 2024-02-10:
> > Hey @Korsberg, this has been mentioned in #72 and should be fixed with #73
> 
> does it mean death of this addon, project closure ?

Don't wanna hulp to conclusions but it does seem like a longer time than usual for the owner to respond.
Definitely open to suggestions.

bladehstream, 2024-02-23:
I am really worried that this is the end, really hoping someone can pick it up.

orkenspalter, 2024-02-26:
I‘m hoping, too.